sunfounder/SunFounder_PiCar-V #1: Doesn't work

Nuruddinjr, 2016-09-08:
I installed v1 of the software and it works but v2 doesn't work on Raspberry pi 3

sunfounder, 2016-09-08:
Sorry about that, V2 is a new better version which is still on the road, and will be release soon. But the code is not compatible with V1

sunfounder, 2016-09-09:
Dear friend,

Sorry for the late response. We have updated the code on github. Please download the new code to try if the car can work well. 

Best regards,

Jake

support@sunfounder.com
 
From: Nuruddin Iminokhunov
Date: 2016-09-08 15:12
To: sunfounder/SunFounder_Smart_Video_Car_Kit_V2.0_for_Raspberry_Pi
Subject: [sunfounder/SunFounder_Smart_Video_Car_Kit_V2.0_for_Raspberry_Pi] Doesn't work (#1)
I installed v1 of the software and it works but v2 doesn't work on Raspberry pi 3

closed 2016-09-19 by sunfounder