sunfounder/SunFounder_PiCar-V #2: Hi Guys

Maz86, 2017-06-09:
I am receiving this error when i am running the server.  please advise

 SunFounder_PiCar  SunFounder_PiCar-V
pi@raspberrypi:~/git $ cd ~/git/SunFounder_PiCar-V/remote_control
pi@raspberrypi:~/git/SunFounder_PiCar-V/remote_control $ sudo ./start

Server running
Performing system checks...

Unhandled exception in thread started by <function wrapper at 0x75f3f230>
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/django/utils/autoreload.py", line 227, in wrapper
    fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/django/core/management/commands/runserver.py", line 125, in inner_run
    self.check(display_num_errors=True)
  File "/usr/local/lib/python2.7/dist-packages/django/core/management/base.py", line 359, in check
    include_deployment_checks=include_deployment_checks,
  File "/usr/local/lib/python2.7/dist-packages/django/core/management/base.py", line 346, in _run_checks
    return checks.run_checks(**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/django/core/checks/registry.py", line 81, in run_checks
    new_errors = check(app_configs=app_configs)
  File "/usr/local/lib/python2.7/dist-packages/django/core/checks/urls.py", line 16, in check_url_config
    return check_resolver(resolver)
  File "/usr/local/lib/python2.7/dist-packages/django/core/checks/urls.py", line 26, in check_resolver
    return check_method()
  File "/usr/local/lib/python2.7/dist-packages/django/urls/resolvers.py", line 254, in check
    for pattern in self.url_patterns:
  File "/usr/local/lib/python2.7/dist-packages/django/utils/functional.py", line 35, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/usr/local/lib/python2.7/dist-packages/django/urls/resolvers.py", line 405, in url_patterns
    patterns = getattr(self.urlconf_module, "urlpatterns", self.urlconf_module)
  File "/usr/local/lib/python2.7/dist-packages/django/utils/functional.py", line 35, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/usr/local/lib/python2.7/dist-packages/django/urls/resolvers.py", line 398, in urlconf_module
    return import_module(self.urlconf_name)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/pi/git/SunFounder_PiCar-V/remote_control/remote_control/urls.py", line 18, in <module>
    import views, settings
  File "/home/pi/git/SunFounder_PiCar-V/remote_control/remote_control/views.py", line 19, in <module>
    picar.setup()
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 49, in setup

  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/SunFounder_PCA9685/PCA9685.py", line 110, in __init__

IOError: [Errno 2] No such file or directory

1AB12, 2017-06-11:
Yes, well ... it looks like you have a long list of errors though python 2.7 keeps reappearing. Have you tried the latest version of python?

Maz86, 2017-06-11:
No, but this is a good shout. I have the pi which is the server ruining on 2.7 and my laptop (client) on 3.6. I'm sure I read somewhere that the pi server should be running on python 2.7. Will give this a try though

Maz86, 2017-06-12:
Fixed this. By enabling ic2 on the pi.

Maz86, 2017-06-12:
resolved #2

sunfounder, 2017-06-14:
Yes you need to enable the I2C, so im gonna close this issus

closed 2017-06-14 by sunfounder