sunfounder/SunFounder_PiCar-V #3: Sun Founder PiCar-V File "manage.py"

1AB12, 2017-06-11:
In the Raspberry Pi terminal:
**cd ~/SunFounder_PiCar-V/remote_control/** (this is fine - no problem)

**sudo ./start** (this has stopped working and produces the following error message -

**Server running
Traceback (most recent call last):
File "manage.py", line 8, in 
from django.core.management import execute_from_command_line
ImportError: No module named django.core.management**

NB: I've tried starting again and installing everything again including:

- reinstalling the the Raspbian operating system (Raspbian is fine)
- reinstalling Python 3.6.1 (I've checked this is ok)
- Reinstalling PyQt5 (I've checked this is installed ok)
- Re-downloading and unzipping from https://github.com/sunfounder/SunFounder_PiCar-V
- (I've also tried sudo apt-get install git and then clone the code from git clone https://github.com/sunfounder/SunFounder_PiCar-V.git)

Despite all of the above the error message continues to appear!?
.... I'm left wondering if SunFounder_PiCar-V is located in the right place. At the moment I've located it in the Python3.6.1 folder. Apart from this I'm bereft of ideas of how to proceed.

Does anyone have a/the solution?

Maz86, 2017-06-11:
Hi is your solution working now? Do you mind casting your eye over my bug as it's similar to yours. However i did install dependencies, can't understand. 

https://github.com/sunfounder/SunFounder_PiCar-V/issues/2

sunfounder, 2017-06-14:
@1AB12 I dont understand how you locate the repo in python 3.6.1. but FYI this script shoud be running under Python2.7,

hannesa2, 2021-01-01:
> this script shoud be running under Python2.7,

@sunfounder 
I guess this is no more a valid statement, am I right ?
All is now `python3` ?

Please see 1944d9a678e1 be8788ae054f316 2cd15e858ca1