sunfounder/SunFounder_PiCar-V #4: picar servo-install Traceback error

jjtronics, 2017-08-18:
Hello,
I have an error on the fresh install of Rapbian : 

pi@raspberrypi:~/SunFounder_PiCar $ picar servo-install
Traceback (most recent call last):
  File "/usr/local/bin/picar", line 11, in <module>
    load_entry_point('SunFounder-PiCar==1.0.0', 'console_scripts', 'picar')()
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 15, in main
    servo0 = Servo(0, bus_number=1)
TypeError: 'module' object is not callable

Another command of picar found : 
pi@raspberrypi:~/SunFounder_PiCar $ picar front-wheel-test
DEBUG "front_wheels.py": Set debug off
DEBUG "front_wheels.py": Set wheel debug off
DEBUG "Servo.py": Set debug off
turn_left
turn_straight
...

picar rear-wheel-test
DEBUG "back_wheels.py": Set debug off
DEBUG "TB6612.py": Set debug off
DEBUG "TB6612.py": Set debug off
DEBUG "PCA9685.py": Set debug off
Forward, speed = 0
Forward, speed = 1
Forward, speed = 2
Forward, speed = 3
...

Can you help me please.

Thank you in advance,

jjtronics, 2017-08-19:
And i have a problm with : 
pi@raspberrypi:~/SunFounder_PiCar-V $ sudo python i2cHelper.py

   ====================================
   ||                                ||
   ||     Raspberry Pi I2C check     ||
   ||                                ||
   ====================================

   Checking your Pi's information.
Traceback (most recent call last):
  File "i2cHelper.py", line 146, in <module>
    main()
  File "i2cHelper.py", line 87, in main
    you_i2c_bus_number = getPiI2CBusNumber()
  File "i2cHelper.py", line 45, in getPiI2CBusNumber
    raise ValueError("Error occur while getting Pi Revision. Revision:{0}".format(revision))
ValueError: Error occur while getting Pi Revision. Revision:2a02082

Thank you in advance,

jjtronics, 2017-08-19:
In i2cHelper.py, i change : 
RPI_REVISION_3 = [2a02082", "a22082"]
to
RPI_REVISION_3 = ["2a02082", "2a22082"]
And the script found my devices : 

pi@raspberrypi:~/SunFounder_PiCar-V $ sudo python i2cHelper.py

   ====================================
   ||                                ||
   ||     Raspberry Pi I2C check     ||
   ||                                ||
   ====================================

   Checking your Pi's information.
   Your cpu revision: 2a02082
   Your Raspberry Pi is Revision 3
   Your I2C bus number is: 1

   Checking your device...
   I2C setting is fine.
   Runing i2cdetect..

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --

Thank you in advance,

jjtronics, 2017-08-19:
And i can etablish a connexion with a server but the control not found : 
[19/Aug/2017 01:18:35] "GET /turning/138 HTTP/1.1" 404 2406
[19/Aug/2017 01:18:35] "GET /motor/stop/ HTTP/1.1" 404 2406
Not Found: /turning/203
[19/Aug/2017 01:18:35] "GET /turning/203 HTTP/1.1" 404 2406
Not Found: /turning/139
[19/Aug/2017 01:18:35] "GET /turning/139 HTTP/1.1" 404 2406
Not Found: /motor/stop/
[19/Aug/2017 01:18:35] "GET /motor/stop/ HTTP/1.1" 404 2406
Not Found: /motor/stop/
Not Found: /turning/203
Not Found: /motor/stop/
Not Found: /turning/138

Thank you in advance,

sunfounder, 2017-08-25:
Hi @jjtronics 
Sorry for being late. the picar issus is fix by newest commit of PiCar. 
To update:

    cd /home/pi/SunFounder_PiCar
    sudo git pull origin master
    sudo python setup.py install

the Server problem, whitch client did you run? the ip:8000 webpage? or the Android App?
The Android App is only for Smart viceo Car Kit the Raspberry Pi Car V1. not PiCar - the Raspberry Pi Car V2. Cuz the App is so unstable, and we are not good at android apps. So, instead we provide webpage controls, which is what we good at and also perform well on nearly all kinds of platform.

chandanzambia, 2018-04-02:
Hi @jjtronics ,
   Did you manage to set up everything?are you able to run client .py to control the car.I have the same issue.if you managed plz guide me.

thanking you.

sunfounder, 2018-05-14:
Can you give us the screenshot of your problem？thanks

support@sunfounder.com
 
From: mbalci21
Date: 2018-05-13 09:17
To: sunfounder/SunFounder_PiCar-V
CC: SunFounder; Comment
Subject: Re: [sunfounder/SunFounder_PiCar-V] picar servo-install Traceback error (#4)
I had a similar problem with device id a020d3.
Also I see picar servo-install not terminating (or making any sound), any advise on this?

drmc149, 2019-12-09:
Hey... is there a solution to this?  I'm getting the same issue @sunfounder @jjtronics

drmc149, 2021-03-10:
I wasn't able to get it to work.  I ordered a second car and the same issue happened with that as well.  I tried what worked for other people and it didn't work for me.

Please let me know if you happen to get it to work.  I'd like to know.

petercame, 2021-03-10:
Thanks for your reply.   I am still trying to get it to work but so far not good.    I am hoping someone will reply to us both to help out.    Keep in touch.