sunfounder/SunFounder_PiCar-V #6: install_dependencies fail

thinavila, 2017-12-08:
Hi, 
When I run an install_dependencies command in raspberry pi, I get an error that python set up has failed. Please see the attached screenshot. I am sorry for the image quality, as it was taken off my TV and not off a computer monitor. I am connected to the internet, and the files were downloaded completely. 
[image]
What is going wrong?

ChasWelles, 2017-12-08:
I wonder if the code has a typo..  I am also having a hard time with loading the django.

ythalorossy, 2017-12-09:
Same problem.
I think the problem is because Raspiberry Pi 3 has two version of python.

Fnjn, 2018-04-11:
This is a naming problem -- bash variable name cannot contain dash ('-')

Simple fix: change every 'is_installed_libjpeg8-dev' to 'is_installed_libjpeg8_dev'