sunfounder/SunFounder_PiCar-V #12: picar servo-install: IndentationError: unindent does not match any outer indentation level

mohamedsaif, 2018-05-23:
I get the following error after running (picar servo-install):
IndentationError: unindent does not match any outer indentation level

Kindly advise.

Below is the full output:
pi@raspberrypi:~/SunFounder_PiCar-V $ picar servo-install
Traceback (most recent call last):
  File "/usr/local/bin/picar", line 11, in <module>
    load_entry_point('SunFounder-PiCar==1.0.0', 'console_scripts', 'picar')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 5, in <module>
    import PCF8591
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/PCF8591.py", line 21
    RPI_REVISION_0 = ["900092"]
                              ^
IndentationError: unindent does not match any outer indentation level

Appsdyo, 2018-05-24:
I have the same error.

Please advise @sunfounder

Appsdyo, 2018-05-24:
I received this reply from sunfounder (for another error, but anyway). Maybe it will work as they are asking to try "picar servo-install"

Hi,

Already solved, please try to update PiCar's code:
Cd ~/SunFounder_PiCar
Sudo git pull origin master
Sudo python setup.py install
Then try picar servo-install

Best regards
Jake