sunfounder/SunFounder_PiCar-V #13: Remote_control ./start

Appsdyo, 2018-05-24:
Hi,
When i try to launch the server, I get an error.

Please advise

pi@raspberrypi:~/SunFounder_PiCar-V/remote_control $ sudo ./start
Server running
Performing system checks...

Unhandled exception in thread started by <function wrapper at 0x75a8c1f0>
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/django/utils/autoreload.py", line 228, in wrapper
    fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/django/core/management/commands/runserver.py", line 124, in inner_run
    self.check(display_num_errors=True)
  File "/usr/local/lib/python2.7/dist-packages/django/core/management/base.py", line 359, in check
    include_deployment_checks=include_deployment_checks,
  File "/usr/local/lib/python2.7/dist-packages/django/core/management/base.py", line 346, in _run_checks
    return checks.run_checks(**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/django/core/checks/registry.py", line 81, in run_checks
    new_errors = check(app_configs=app_configs)
  File "/usr/local/lib/python2.7/dist-packages/django/core/checks/urls.py", line 16, in check_url_config
    return check_resolver(resolver)
  File "/usr/local/lib/python2.7/dist-packages/django/core/checks/urls.py", line 26, in check_resolver
    return check_method()
  File "/usr/local/lib/python2.7/dist-packages/django/urls/resolvers.py", line 254, in check
    for pattern in self.url_patterns:
  File "/usr/local/lib/python2.7/dist-packages/django/utils/functional.py", line 35, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/usr/local/lib/python2.7/dist-packages/django/urls/resolvers.py", line 405, in url_patterns
    patterns = getattr(self.urlconf_module, "urlpatterns", self.urlconf_module)
  File "/usr/local/lib/python2.7/dist-packages/django/utils/functional.py", line 35, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/usr/local/lib/python2.7/dist-packages/django/urls/resolvers.py", line 398, in urlconf_module
    return import_module(self.urlconf_name)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/pi/SunFounder_PiCar-V/remote_control/remote_control/urls.py", line 18, in <module>
    import views, settings
  File "/home/pi/SunFounder_PiCar-V/remote_control/remote_control/views.py", line 14, in <module>
    from driver import camera, stream
  File "/home/pi/SunFounder_PiCar-V/remote_control/remote_control/driver/camera.py", line 14, in <module>
    from picar.SunFounder_PCA9685 import Servo
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 5, in <module>
    import PCF8591
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/PCF8591.py", line 21
    RPI_REVISION_0 = ["900092"]
                              ^
IndentationError: unindent does not match any outer indentation level

sunfounder, 2018-05-24:
Hi,

Thanks for contacting us,OK,I have asked our engineers to check your problem,and I have any result.I will reply you.
thanks

Best

Jake

support@sunfounder.com
 
From: Appsdyo
Date: 2018-05-24 10:43
To: sunfounder/SunFounder_PiCar-V
CC: Subscribed
Subject: [sunfounder/SunFounder_PiCar-V] Remote_control ./start (#13)
Hi,
When i try to launch the server, I get an error.
Please advise
pi@raspberrypi:~/SunFounder_PiCar-V/remote_control $ sudo ./start
Server running
Performing system checks...
Unhandled exception in thread started by <function wrapper at 0x75a8c1f0>
Traceback (most recent call last):
File "/usr/local/lib/python2.7/dist-packages/django/utils/autoreload.py", line 228, in wrapper
fn(*args, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/django/core/management/commands/runserver.py", line 124, in inner_run
self.check(display_num_errors=True)
File "/usr/local/lib/python2.7/dist-packages/django/core/management/base.py", line 359, in check
include_deployment_checks=include_deployment_checks,
File "/usr/local/lib/python2.7/dist-packages/django/core/management/base.py", line 346, in _run_checks
return checks.run_checks(**kwargs)
File "/usr/local/lib/python2.7/dist-packages/django/core/checks/registry.py", line 81, in run_checks
new_errors = check(app_configs=app_configs)
File "/usr/local/lib/python2.7/dist-packages/django/core/checks/urls.py", line 16, in check_url_config
return check_resolver(resolver)
File "/usr/local/lib/python2.7/dist-packages/django/core/checks/urls.py", line 26, in check_resolver
return check_method()
File "/usr/local/lib/python2.7/dist-packages/django/urls/resolvers.py", line 254, in check
for pattern in self.url_patterns:
File "/usr/local/lib/python2.7/dist-packages/django/utils/functional.py", line 35, in get
res = instance.dict[self.name] = self.func(instance)
File "/usr/local/lib/python2.7/dist-packages/django/urls/resolvers.py", line 405, in url_patterns
patterns = getattr(self.urlconf_module, "urlpatterns", self.urlconf_module)
File "/usr/local/lib/python2.7/dist-packages/django/utils/functional.py", line 35, in get
res = instance.dict[self.name] = self.func(instance)
File "/usr/local/lib/python2.7/dist-packages/django/urls/resolvers.py", line 398, in urlconf_module
return import_module(self.urlconf_name)
File "/usr/lib/python2.7/importlib/init.py", line 37, in import_module
import(name)
File "/home/pi/SunFounder_PiCar-V/remote_control/remote_control/urls.py", line 18, in 
import views, settings
File "/home/pi/SunFounder_PiCar-V/remote_control/remote_control/views.py", line 14, in 
from driver import camera, stream
File "/home/pi/SunFounder_PiCar-V/remote_control/remote_control/driver/camera.py", line 14, in 
from picar.SunFounder_PCA9685 import Servo
File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/init.py", line 5, in 
import PCF8591
File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/PCF8591.py", line 21
RPI_REVISION_0 = ["900092"]
^
IndentationError: unindent does not match any outer indentation level

Appsdyo, 2018-05-24:
Thanks. Since Sunday I am struggling with code errors.

Right now, the car is still not functionning (and useless).

sunfounder, 2018-05-24:
Hi,

Already solved, please try to update PiCar's code:
Cd ~/SunFounder_PiCar
Sudo git pull origin master
Sudo python setup.py install
Then try picar servo-install

Best regards
Jake

support@sunfounder.com
 
From: Appsdyo
Date: 2018-05-24 11:38
To: sunfounder/SunFounder_PiCar-V
CC: SunFounder; Comment
Subject: Re: [sunfounder/SunFounder_PiCar-V] Remote_control ./start (#13)
Thanks. Since Sunday I am struggling with code errors.
Right now, the car is still not functionning (and useless).

Appsdyo, 2018-05-24:
Thanks for your reply. I will check later.

Nevertheless, the error I am facing is with the file "start" located in "remote control", and not the "picar servo-install" (By the way, I am getting an error with this one as informed in another post).

Anyway, I will let you know.

Appsdyo, 2018-05-25:
Yes it is working perfectly ! 
Thanks.

closed 2018-05-25 by Appsdyo