sunfounder/SunFounder_PiCar-V #15: IOError: [Errno 2] No such file or directory

prova4, 2018-06-25:
we tried to build the car, but when we ran the command "picar servo-install" it would give us an error.
Traceback (most recent call last):
  File "/usr/local/bin/picar", line 11, in <module>
    load_entry_point('SunFounder-PiCar==1.0.0', 'console_scripts', 'picar')()
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 43, in main
    setup()
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 80, in setup
    pwm=PCA9685.PWM(bus_number=1)
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/SunFounder_PCA9685/PCA9685.py", line 116, in __init__
    self.bus = smbus.SMBus(self.bus_number)
IOError: [Errno 2] No such file or directory

prova4, 2018-06-25:
### We also tried to run "sudo ./show" but it gave us this error.
sudo bash show 
show: line 2: import: command not found
show: line 4: syntax error near unexpected token `('
show: line 4: `license = open('LICENSE')'

sunfounder, 2018-06-26:
Hi dear 

Have your problem been solved?If you haven't ,you can email me.thanks

Best Regards
Jake 

support@sunfounder.com
 
From: prova4
Date: 2018-06-26 00:24
To: sunfounder/SunFounder_PiCar-V
CC: Subscribed
Subject: Re: [sunfounder/SunFounder_PiCar-V] IOError: [Errno 2] No such file or directory (#15)
We managed to fix the problem
the problem is i2c
https://unix.stackexchange.com/questions/106886/piglow-not-working

closed 2018-06-29 by prova4

sunfounder, 2018-07-02:
Hi prova4

So sorry,I don't understand  what you said,can you give me detailed information.thanks

Best Regards
Jake 

support@sunfounder.com
 
From: prova4
Date: 2018-06-29 17:00
To: sunfounder/SunFounder_PiCar-V
CC: SunFounder; Comment
Subject: Re: [sunfounder/SunFounder_PiCar-V] IOError: [Errno 2] No such file or directory (#15)
Closed #15.