sunfounder/SunFounder_PiCar-V #16: Remote control connection hangs and freezes

samueljmorris, 2018-07-24:
When I connect to the PiCar via WiFi, the client periodically hangs, the video freezes, and the car will continue whatever command it received last. Often, this results in the PiCar crashing into an object.

The issue is the most common using the Python client on Windows.

Using the web interface instead of the Python client helps, but it still freezes from time to time.

I have tried using both the Raspberry Pi 3 B onboard wifi and the supplied Ralink adapter.

The issue occurs even if the WiFi link quality shows as 70/70 in ifconfig.

When it is frozen, the Pi itself is still working normally but the output of the server process does not show any new commands being received.

sunfounder, 2018-07-24:
Hi samueljmorris

Your question, I have sent feedback to our engineers to detect, if they give any results, I will inform you immediately, if there are other questions, please contact us anytime, we will try our best to solve it for you.

Best Regards
Jake 

support@sunfounder.com
 
From: samueljmorris
Date: 2018-07-24 10:20
To: sunfounder/SunFounder_PiCar-V
CC: Subscribed
Subject: [sunfounder/SunFounder_PiCar-V] Remote control connection hangs and freezes (#16)
When I connect to the PiCar via WiFi, the client periodically hangs, the video freezes, and the car will continue whatever command it received last. Often, this results in the PiCar crashing into an object.
The issue is the most common using the Python client on Windows.
Using the web interface instead of the Python client helps, but it still freezes from time to time.
I have tried using both the Raspberry Pi 3 B onboard wifi and the supplied Ralink adapter.
The issue occurs even if the WiFi link quality shows as 70/70 in ifconfig.
When it is frozen, the Pi itself is still working normally but the output of the server process does not show any new commands being received.

samueljmorris, 2018-07-24:
Thank you for following up! I have looked into the issue more on my end, and it seems to occur when the WiFi signal strength drops too low and packet errors start occurring. Wireshark shows lots of duplicate ACK packets, out of order packets, and spurious retransmissions when it hangs.

Is there a way to use some kind of watchdog that will stop the car's motion if the connection times out?

My main concern is that the car keeps moving in whatever direction it was last going when the connection hangs.