sunfounder/SunFounder_PiCar-V #17: Client script not working

ujjwalrathod007, 2018-08-13:
I have issues runnig the client on of the Robocar.

On windows I get following error 

When I do python client.py I get the UI when I Log in

ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it. 

On ubuntu there is problem with PyQt5. "QtCore.so: undefined symbol "

What should I do?

sunfounder, 2018-08-16:
Hi

Please send us the results of the raspberry pi operation.

Best Regards
Jake 

support@sunfounder.com
 
From: support@sunfounder.com
Date: 2018-08-14 21:20
To: sunfounder/SunFounder_PiCar-V
Subject: Re: [sunfounder/SunFounder_PiCar-V] Client script not working (#17)
Hi ,
Thanks for contacting us.Your question, I have asked our engineers help , if they give any results, I will inform you immediately, 
if there are other questions, please contact us anytime, we will try our best to solve it for you. 

Best Regards
Jake 

support@sunfounder.com
 
From: Ujjval Rathod
Date: 2018-08-13 23:51
To: sunfounder/SunFounder_PiCar-V
CC: Subscribed
Subject: [sunfounder/SunFounder_PiCar-V] Client script not working (#17)
I have issues runnig the client on of the Robocar.
On windows I get following error
When I do python client.py I get the UI when I Log in
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it.
On ubuntu there is problem with PyQt5. "QtCore.so: undefined symbol "
What should I do?

ujjwalrathod007, 2018-08-17:
Hi,

On My RPI I start the server as shown in the guide. it runs fine as given I
guess.

BR,
Ujjval Rathod.

sunfounder, 2018-08-22:
Hi dear

That is GREAT.

Best Regards
Jake 

support@sunfounder.com
 
From: Ujjval Rathod
Date: 2018-08-17 15:14
To: sunfounder/SunFounder_PiCar-V
CC: SunFounder; Comment
Subject: Re: [sunfounder/SunFounder_PiCar-V] Client script not working (#17)
Hi,

On My RPI I start the server as shown in the guide. it runs fine as given I
guess.

BR,
Ujjval Rathod.

ujjwalrathod007, 2018-08-22:
Hi,

Yes but what about the client. the errors I have is in starting the client.

Best Regards,
Ujjval Rathod.

sunfounder, 2018-08-30:
Hi dear

Is it client software that run in Windows and Linux  , we all have our tutorial: https://www.sunfounder.com/learn/download/U21hcnRfVmlkZW9fQ2FyX1YyLjBfZm9yX1Jhc3BiZXJyeV9QaV9QaUNhci1WXy5wZGY=/dispi, could you tell me how you can not run, can you send us the picture?Thanks

Best Regards
Jake 

support@sunfounder.com
 
From: Ujjval Rathod
Date: 2018-08-22 22:59
To: sunfounder/SunFounder_PiCar-V
CC: SunFounder; Comment
Subject: Re: [sunfounder/SunFounder_PiCar-V] Client script not working (#17)
Hi,

Yes but what about the client. the errors I have is in starting the client.

Best Regards,
Ujjval Rathod.

juangascon, 2018-10-11:
To solve this issue, in a virtualenv, I upgraded PyQt5 to the latest version: 5.11.3 and installed also 'requests' 
This worked for me.

``` bash
pip3 install --upgrade pyqt5 requests
```

This is the content of my requirements.txt file for 'pip3 install -r requirements.txt' :

```
certifi==2018.8.24
chardet==3.0.4
idna==2.7
pkg-resources==0.0.0
PyQt5==5.11.3
PyQt5-sip==4.19.12
requests==2.19.1
urllib3==1.23
```

Good luck