sunfounder/SunFounder_PiCar-V #18: picar servo-install command fails

mikhalchuk, 2018-08-30:
I have an issue with a `picar servo-install` command, here is a stack trace:
```bash
$ picar servo-install
Traceback (most recent call last):
  File "/usr/local/bin/picar", line 11, in <module>
    load_entry_point('SunFounder-PiCar==1.0.0', 'console_scripts', 'picar')()
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 43, in main
    setup()
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 80, in setup
    pwm=PCA9685.PWM(bus_number=1)
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/SunFounder_PCA9685/PCA9685.py", line 116, in __init__
    self.bus = smbus.SMBus(self.bus_number)
IOError: [Errno 2] No such file or directory
```

Please advise @sunfounder

sunfounder, 2018-08-31:
Hi dear

The smbus is not configured, detailed tutorial about the i2c configuration，please refer to our website wiki，
http://wiki.sunfounder.cc/index.php? Title = I2C_Configuration
Thanks so much.

Best Regards
Jake 

support@sunfounder.com
 
From: Sergii
Date: 2018-08-31 04:15
To: sunfounder/SunFounder_PiCar-V
CC: SunFounder; Mention
Subject: [sunfounder/SunFounder_PiCar-V] picar servo-install command fails (#18)
I have an issue with a picar servo-install command, here is a stack trace:
$ picar servo-install
Traceback (most recent call last):
  File "/usr/local/bin/picar", line 11, in <module>
    load_entry_point('SunFounder-PiCar==1.0.0', 'console_scripts', 'picar')()
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 43, in main
    setup()
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 80, in setup
    pwm=PCA9685.PWM(bus_number=1)
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/SunFounder_PCA9685/PCA9685.py", line 116, in __init__
    self.bus = smbus.SMBus(self.bus_number)
IOError: [Errno 2] No such file or directory
Please advise @sunfounder

sunfounder, 2018-09-04:
Hi dear

So sory for reply you late
Specific reason ,please refer to our WIKI: http://wiki.sunfounder.cc/index.php? Title = I2C_Configuration

Thanks 

Best Regards
Jake 

support@sunfounder.com
 
From: Sergii
Date: 2018-08-31 04:15
To: sunfounder/SunFounder_PiCar-V
CC: SunFounder; Mention
Subject: [sunfounder/SunFounder_PiCar-V] picar servo-install command fails (#18)
I have an issue with a picar servo-install command, here is a stack trace:
$ picar servo-install
Traceback (most recent call last):
  File "/usr/local/bin/picar", line 11, in <module>
    load_entry_point('SunFounder-PiCar==1.0.0', 'console_scripts', 'picar')()
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 43, in main
    setup()
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/__init__.py", line 80, in setup
    pwm=PCA9685.PWM(bus_number=1)
  File "/usr/local/lib/python2.7/dist-packages/SunFounder_PiCar-1.0.0-py2.7.egg/picar/SunFounder_PCA9685/PCA9685.py", line 116, in __init__
    self.bus = smbus.SMBus(self.bus_number)
IOError: [Errno 2] No such file or directory
Please advise @sunfounder