sunfounder/SunFounder_PiCar-V #27: Run the server automatically

Llassadar, 2018-12-29:
Hello,

is there the possibility, that the server is run automatically when i start the raspi, so that i do not need to log into the Raspi remotely every time when i want to drive the car?

Jmlannan, 2019-03-18:
See here for running a program at start, see systemd part
[https://www.dexterindustries.com/howto/run-a-program-on-your-raspberry-pi-at-startup/#systemd](https://www.dexterindustries.com/howto/run-a-program-on-your-raspberry-pi-at-startup/#systemd)

jpmc3630, 2020-11-28:
In case anyone else comes along looking for this..
Just add a few lines to the rc.local script (found at /etc/rc.local) which runs at start up.

To use the nano text editor run command:
`sudo nano /etc/rc.local`

Then add at the bottom:
`cd /home/pi/car/remote_control
./start`

Gaploid, 2021-02-14:
@jpmc3630 it seems that option does not work. It's not constantly work in the background and exit instantly.