sunfounder/SunFounder_PiCar-V #30: missing parts in kit, cad model ?

EmFl, 2019-04-10:
Hello,
I have bought a picar v with missing parts preventing me to assemble the front wheels. Missing parts are  6 bearing shields and 6 Hex front wheel fixing plate. I reached out to service@sunfounder.com but for now didn't get any reply. Since those are small basic parts I was wondering if someone could give me their specs so that I can use a 3d printer to make them ? 

Thanks

sunfounder, 2019-04-10:
Dear customer,
Thank you for getting in touch,we are sorry to hear about that,to better assist you ,could you send us your order number and order screenshot? our other colleagues will help you to deal with it.Thanks for your information in advance.

Best Regards,
SunFounder Team

support@sunfounder.com
 
From: EmFl
Date: 2019-04-10 10:44
To: sunfounder/SunFounder_PiCar-V
CC: Subscribed
Subject: [sunfounder/SunFounder_PiCar-V] missing parts in kit, cad model ? (#30)
Hello,
I have bought a picar v with missing parts preventing me to assemble the front wheels. Missing parts are 6 bearing shields and 6 Hex front wheel fixing plate. I reached out to service@sunfounder.com but for now didn't get any reply. Since those are small basic parts I was wondering if someone could give me their specs so that I can use a 3d printer to make them ?
Thanks

EmFl, 2019-04-10:
Thanks, sent an email, with order details. (Sorry to use github for this)

Theosakamg, 2021-02-21:
You can find more improve (3D printer) at :
https://github.com/Theosakamg/PiCar_Hardware

@EmFl : can you share the original CAD file ?