sunfounder/SunFounder_PiCar-V #31: Servo install syntax error

MJimmy, 2019-04-19:
I have a problem wwith servo installing. I followed the instructions by step by step. After I wrote the command "picar servo-install" it says SyntaxError : Missing parentheses in call to ''print''
pls help 
[image]

sunfounder, 2019-04-22:
Dear customer,
Thank you for contacting us concerning your question.
We have re-updated the code to the git,please kindly check that.
Please delect the sunfounderpicar and sunfounderpicarv, excute the script again after re-downloading. 
If you have further questions,please feel free to contact us.

Best Regards,
SunFounder Team
 
From: MJimmy
Date: 2019-04-19 17:18
To: sunfounder/SunFounder_PiCar-V
CC: Subscribed
Subject: [sunfounder/SunFounder_PiCar-V] Servo install syntax error (#31)
I have a problem wwith servo installing. I followed the instructions by step by step. After I wrote the command "picar servo-install" it says SyntaxError : Missing parentheses in call to ''print''
pls help

5ocjal, 2019-08-23:
The same error here. I've delete both SunFounder dirs, cloned repo and run the script once again, but when I type "picar servo-install" I got that:

`pi@raspberrypi:~/SunFounder_PiCar $ sudo picar servo-install
Traceback (most recent call last):
  File "/usr/local/bin/picar", line 11, in <module>
    load_entry_point('SunFounder-PiCar==1.0.1', 'console_scripts', 'picar')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.7/dist-packages/SunFounder_PiCar-1.0.1-py3.7.egg/picar/__init__.py", line 2, in <module>
    from . import front_wheels
  File "/usr/local/lib/python3.7/dist-packages/SunFounder_PiCar-1.0.1-py3.7.egg/picar/front_wheels.py", line 14, in <module>
    from .SunFounder_PCA9685 import Servo
  File "/usr/local/lib/python3.7/dist-packages/SunFounder_PiCar-1.0.1-py3.7.egg/picar/SunFounder_PCA9685/Servo.py", line 15, in <module>
    from . import PCA9685
  File "/usr/local/lib/python3.7/dist-packages/SunFounder_PiCar-1.0.1-py3.7.egg/picar/SunFounder_PCA9685/PCA9685.py", line 14, in <module>
    import smbus
ModuleNotFoundError: No module named 'smbus'`

Ruloch-Ranperoge, 2019-12-04:
First, install `smbus` by
```
pip3 install smbus
```
Then, enable I2C of raspberry pi by
```
sudo raspi-config
```