sunfounder/SunFounder_PiCar-V #37: remote_control ./start "You have 1 unapplied migration(s)." issue

DanielKoohmarey, 2019-10-06:
When running the remote_control ./start on a new Raspberry Pi 3+ you get:
```
pi@raspberrypi:~/SunFounder_PiCar-V/remote_control $ sudo ./start
Server running
Performing system checks...

DEBUG "front_wheels.py": Set debug off
DEBUG "front_wheels.py": Set wheel debug off
DEBUG "Servo.py": Set debug off
DEBUG "back_wheels.py": Set debug off
DEBUG "TB6612.py": Set debug off
DEBUG "TB6612.py": Set debug off
DEBUG "PCA9685.py": Set debug off
DEBUG "camera.py": Set debug off
DEBUG "camera.py": Set pan servo and tilt servo debug off
DEBUG "Servo.py": Set debug off
DEBUG "Servo.py": Set debug off
mjpg_streamer -i "/usr/local/lib/input_uvc.so" -o "/usr/local/lib/output_http.so                    -w /usr/local/www" &
None
System check identified no issues (0 silenced).

You have 1 unapplied migration(s). Your project may not work properly until you                    apply the migrations for app(s): auth.
Run 'python manage.py migrate' to apply them.
```
The problem is the "Run 'python manage.py migrate' to apply them." output, as the latest Raspbian image comes with both python2 and python3 installed, with 'python' launching python2. As a result to fix the migration issue, you actually have to run `python3 manage.py migrate`.

Eisteepropper, 2020-02-12:
how to run python3 manage.py migrate?
i'm sorry this might be a stupid question but i'm very noob in this.

Roya-myob, 2020-05-03:
Just run this command: 'python3 manage.py migrate'.