sunfounder/SunFounder_PiCar-V #38: Front Wheels not moving. Back wheels work fine

Surbhimangla, 2019-10-13:
Hi

I just assembled the PiCar S and installed the software. When i run " picar front-wheel-test" nothing happens. However, the back wheels are mving with the command "picar rear-wheel-test". Please help

selvadurai, 2020-11-11:
Same here, having the same issue.

axi0m, 2020-11-23:
I have an issue with both front and rear wheels not functioning.

hannesa2, 2021-01-03:
Same here, having the same issue.

After `$ picar servo-install` I see
> Servo now is set to 90 degree.

[image]

Then I start server with `cd remote_control/ && sudo ./start `

```
Server running
Performing system checks...

DEBUG "front_wheels.py": Set debug off
DEBUG "front_wheels.py": Set wheel debug off
DEBUG "Servo.py": Set debug off
DEBUG "back_wheels.py": Set debug off
DEBUG "TB6612.py": Set debug off
DEBUG "TB6612.py": Set debug off
DEBUG "PCA9685.py": Set debug off
DEBUG "camera.py": Set debug off
DEBUG "camera.py": Set pan servo and tilt servo debug off
DEBUG "Servo.py": Set debug off
DEBUG "Servo.py": Set debug off
mjpg_streamer -i "/usr/local/lib/input_uvc.so" -o "/usr/local/lib/output_http.so -w /usr/local/www" &
None
System check identified no issues (0 silenced).
January 03, 2021 - 13:47:08
Django version 2.0, using settings 'remote_control.settings'
Starting development server at http://0.0.0.0:8000/
Quit the server with CONTROL-C.
```
Back wheels are working but front wheels do nothing with `A` and `D`. Server receives commands !

[image]

@axi0m @Surbhimangla @selvadurai 
Did you fixed this ?

hannesa2, 2021-01-06:
Something works somehow
```
servoFrontWheel = Servo.Servo(0, bus_number=1)
servoFrontWheel.write(i)
```

works only for values from 180 (almost strait forward) and 135 (45° to the left) 
All values < 135 and > 180 have no affect, and I'm not able to move wheels to the right at all

hannesa2, 2021-01-06:
@sunfounder Please can you help to fix this ?

hannesa2, 2021-01-10:
I found the reason:
Disable front wheels to have the state  https://github.com/sunfounder/SunFounder_PiCar-V/blob/V3.0/docs/(EN)%20PiCar-V%20User%20Manual.pdf p.43 
Do `picar servo-install` this will move your servo the default position, then continue with assemble.....

Now it works properly