sunfounder/SunFounder_PiCar-V #42: Client crashing on 'camready'

parkershamblin, 2019-11-18:
When I click the login button in the client the client crashes. In the console it shows the last command it tried running is 'action=camready'. I've tried running it several time and every time it crashes on 'camready'.

I'm unsure what to do. When I ran the command 'picar servo-install' the console just showed 'Servo now is set to 90 degree.' but nothing else showed up and it appeared as if something was still running. Maybe I not wait long enough during this step and something else is suppose to show up?

Tetrakis, 2020-03-15:
I'm having the same problem.  Additionally, with the Web version of the client I can control the car with the keyboard, but no camera images are displayed.  Page itself is also very off with buttons not looking like they can be pressed, not doing anything if they are and running of the screen.

wbwxshao, 2021-04-01:
Hello, I have the same issue as Client crashing on 'camready'. Did you find any solution to this?

QRMAT, 2021-04-24:
Hello, just set the port to 8080 (same as mjpg_streamer uses) solves this for me. Perhaps also check the firewall ports for blocking the port.