sunfounder/SunFounder_PiCar-V #43: "Connection refused"

parkershamblin, 2019-11-20:
Not sure what's wrong. I've followed the instructions exactly. Here's what I see when I have the server up and running and try to log into the client.

`url: http://192.168.2.254:8000/run/?action=camready
Traceback (most recent call last):
  File "client.py", line 288, in reflash_frame
    pixmap = self.transToPixmap()
  File "client.py", line 271, in transToPixmap
    data = self.queryImage.queryImage()
  File "client.py", line 689, in queryImage
    http_data.endheaders()
  File "/usr/lib/python3.7/http/client.py", line 1224, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1016, in _send_output
    self.send(msg)
  File "/usr/lib/python3.7/http/client.py", line 956, in send
    self.connect()
  File "/usr/lib/python3.7/http/client.py", line 928, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "/usr/lib/python3.7/socket.py", line 727, in create_connection
    raise err
  File "/usr/lib/python3.7/socket.py", line 716, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
Aborted
`

AUGSLAG, 2022-09-25:
Same issue. No response.

[image]

parkershamblin, 2023-04-13:
Included my IP Address.

🤦🤦🤦

closed 2023-04-13 by parkershamblin

fabriciofsantana, 2024-03-30:
Hi @parkershamblin , I've got the same error...
Do you mind telling me what you did to resolve it?
Thanks.

AUGSLAG, 2024-03-31:
Gave up.
Used the MIT APP  inventory to program from a raspberry pi .

AUGSLAG, 2024-03-31:
The code is just a bunch of copy paste from other girtubs. They can't
update the time issue because they didn't write the code.