sunfounder/SunFounder_PiCar-V #49: pi car camera v2 - error: that port is already in use

Rudemood76, 2020-09-22:
Hi 
need your help.

i had that "unapplied migration(s)" issue that with the python manage.py, so i did the "python3 manage.py migrate" 
that someone suggested here.

then restarted and tried again, now i get an error "error: that port is already in use" and program aborts!

login as: pi
pi@10.0.0.18's password:
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Sep 22 09:26:57 2020
pi@raspberrypi:~ $ ls
 Desktop                             MagPi            SunFounder_PiCar
 Documents                           Music            SunFounder_PiCar-V
 Downloads                           Pictures         Templates
 dragit_installer.py                 Public           Videos
 line_graph-2019-09-16-10-51-29.py  'say hello.sb2'
pi@raspberrypi:~ $ cd ~/SunFounder_PiCar-V/remote_control
pi@raspberrypi:~/SunFounder_PiCar-V/remote_control $ sudo ./start
Server running
Performing system checks...

DEBUG "front_wheels.py": Set debug off
DEBUG "front_wheels.py": Set wheel debug off
DEBUG "Servo.py": Set debug off
DEBUG "back_wheels.py": Set debug off
DEBUG "TB6612.py": Set debug off
DEBUG "TB6612.py": Set debug off
DEBUG "PCA9685.py": Set debug off
DEBUG "camera.py": Set debug off
DEBUG "camera.py": Set pan servo and tilt servo debug off
DEBUG "Servo.py": Set debug off
DEBUG "Servo.py": Set debug off
mjpg_streamer -i "/usr/local/lib/input_uvc.so" -o "/usr/local/lib/output_http.so -w /usr/local/www" &
None
System check identified no issues (0 silenced).
September 22, 2020 - 06:33:13
Django version 2.0, using settings 'remote_control.settings'
Starting development server at http://0.0.0.0:8000/
Quit the server with CONTROL-C.
Error: That port is already in use.
pi@raspberrypi:~/SunFounder_PiCar-V/remote_control $
pi@raspberrypi:~/SunFounder_PiCar-V/remote_control 

any help will be great.
btw, i used to have 2 ip's (???) 10.0.0.18 & 10.0.0.19 but now suddenly the "19" dosent work, only the "18" works.

thanks!

TNTMaster5858, 2021-01-01:
Ya, the same thing is happening to me, does anyone know how to fix this?

Gaploid, 2021-01-09:
I had the same issue after install DragIt environment. It seems remote_control app and DragIt working on the same port - 8000. I just changed that file on my raspberryPi '''
```
~/SunFounder_picar-v/remote_control/start 
```
8000 -> 9000 
and start that server again.

amineremache, 2022-01-01:
What is hapening is that, you run the server once, and when you closed it, it didn't free the 8000 port, or simply that another application is running on that port.
What you can do is either :
- Change the port as mentioned by @Gaploid 
OR 
- Kill the previous process that is still holding the 8000 port by running `sudo lsof -i:8000`, you will have the PID ( or Process ID ), then take that PID and use it with `kill` command as follows : `sudo kill -9 <PID>`

gccheng, 2022-06-25:
> 

Hi Gaploid, which file should we change the port from 8000 to 9000? ~/SunFounder_picar-v/remote_control/start  seems a binary file. Thanks.

sunfounder, 2022-06-27:
Hi sir ,
Thanks for contacting us.
Could you please provide us your order number first? 
Which shopping platform did you get it?
A short video or some pictures of the issues could be much more helpful.
Our staff will contact you later to help with that.
 
Best Regards,

***@***.***
 
From: gccheng
Date: 2022-06-25 12:08
To: sunfounder/SunFounder_PiCar-V
CC: Subscribed
Subject: Re: [sunfounder/SunFounder_PiCar-V] pi car camera v2 - error: that port is already in use (#49)
Hi Gaploid, which file should we change the port from 8000 to 9000? ~/SunFounder_picar-v/remote_control/start seems a binary file. Thanks.