sunfounder/SunFounder_PiCar-V #52: picar servo-install hangs

axi0m, 2020-11-17:
It seems that the version of picar I have sits in an infinite loop. 

File: `/usr/local/lib/python3.5/dist-packages/SunFounder_PiCar-1.0.1-py3.5.egg/picar/__init__.py`

```python
def servo_install():
    import time
    delay = 1.0/180
    if len(sys.argv) >= 3:
        print("servo-install takes no value")
        usage()
    print("Servo now is set to 90 degree.")
    servo0 = Servo.Servo(0, bus_number=1)
    servo1 = Servo.Servo(1, bus_number=1)
    servo2 = Servo.Servo(2, bus_number=1)
    for i in range(90, -1, -1):
        servo0.write(i)
        servo1.write(i)
        servo2.write(i)
        time.sleep(delay)
    time.sleep(0.1)
    for i in range(0, 181, 1):
        servo0.write(i)
        servo1.write(i)
        servo2.write(i)
        time.sleep(delay)
    time.sleep(0.1)
    for i in range(180, 89, -1):
        servo0.write(i)
        servo1.write(i)
        servo2.write(i)
        time.sleep(delay)
    time.sleep(0.1)
    servo0.write(90)
    servo1.write(90)
    servo2.write(90)
    while True:
        time.sleep(1)
```

Notice the while True loop. Thus when I run the command `picar servo-install` it just sits there forever, I don't think I need to keep the python programming running after I've run the command successfully?

Thanks!

axi0m, 2020-11-17:
Build Info
```
pi@raspberrypi:~/SunFounder_PiCar-V/remote_control $ python --version
Python 2.7.13
pi@raspberrypi:~/SunFounder_PiCar-V/remote_control $ python3 --version
Python 3.5.3
pi@raspberrypi:~/SunFounder_PiCar-V/remote_control $ cat /etc/debian_version
9.13
pi@raspberrypi:~/SunFounder_PiCar-V/remote_control $ uname -a
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
```

hannesa2, 2021-01-06:
I made a pull request for this https://github.com/sunfounder/SunFounder_PiCar/pull/10