sunfounder/SunFounder_PiCar-V #61: Just checking wiring diagram - doesn't match physical wires in kit

tbird20d, 2021-01-09:
The wiring diagram on pages 24 and 25, for the motor connections, clearly show the black wire on the left and red wire on the right, in both diagrams.  However, in my kit the connectors have the red wire on the left and black wire on the right, for both motors.  Please let me know if this will be a problem for me.  Thanks.

ctjoumas, 2021-04-24:
I think the instructions are wrong - I was wondering the same thing. To double check, I looked at a youtube video from less than a year ago and it has the red wire on the left and black on the right (and it's the only way it can be plugged in).

Raunak-Singh-Inventor, 2022-11-11:
Thanks thought my motors were defective :)