sunfounder/SunFounder_PiCar-V #62: Camera works, Front wheels works but....rear wheels not work

pabValle, 2021-01-17:
Hello!
I've recently buy the Kit "Smart Video Car Kit 2.0 for Raspberry Pi"
After built the car, install software and the others as i read in the instructions, and finally  run client "client.py" and click to test....i can hear a little continous vibration on the rear wheels motors but they don't work. (don't move)......I haven't seen any obstacle on the motors.....¿Someone else has the same issue?

petercame, 2021-03-04:
using the picar software (eg 'picar rear-wheels-test') I get no response from rear wheels, or from the servo (when using 'servo-install') or from the front wheels when using 'front-wheel-test).  Help, please! Thanks.

petercame, 2021-03-06:
Please can anyone help me with this: I have installed the PiCar software and followed all the set-up procedure; however, when I try the three motor/wheel tests (servo-install, front-wheel-test, rear-wheel-test), although the software responds as if it is has understood the commands, the actual wheels and servo do not respond.

ctjoumas, 2021-04-28:
Sounds silly, but check your batteries. I had the same issue where the server was running and receiving commands just fine, but front wheels and camera wouldn't move and the rear wheels I would hear a very faint vibration sound. My batteries were just fully charged, so I didn't think to look at them, but I finally used a multimeter to test the voltage and one of the batteries was dead (no idea what happened). I replaced that one battery and everything worked. Hopefully it's as simple as that!

petercame, 2021-04-29:
Thanks very much for this.  I will check the batteries.

However,  I had the same problem (steering and drive wheels not working) when the car was connected to mains power.    Am I beng stupid?

ctjoumas, 2021-04-29:
I'm actually not entirely sure, but when I had the car connected just via the power cable to the raspberry pi, the wheels didn't work. I'm not using the official raspberry pi power cable, so it's possible the USB-C cable I'm using isn't supplying enough power; or maybe it's not meant to supply power to the HATS? I know the calibration worked for me, but once it was all fully assembled, the servos never worked with just that power cable. It didn't work until the batteries were in.

I hope that helps and hopefully it's just your batteries. If not, hopefully someone else will be able to advise. service@sunfounder.com has been pretty responsive too just in case it's not the batteries.

petercame, 2021-04-29:
Thank you for your detailed reply.   Everything you describe fits with what I have found.  I will try the batteries (after charging them!) and let you know the result.   (I am away for a few days so will not have time until I'm back home.)

Thanks again.

I hope that helps and hopefully it's just your batteries. If not, hopefully someone else will be able to advise. ***@***.*** has been pretty responsive too just in case it's not the batteries.

ganzx2, 2022-10-04:
Im having the same issue with mine when plugged in. Did the batteries make a difference? I hear a small sound when doing the rear wheel test but nothing during servo install or front wheel test.

petercame, 2022-10-05:
Thanks for your email.    Sorry, I have abandoned the Pi-Car some time ago and can't remember much about it.

ctjoumas, 2022-10-05:
Hi ganzx2,

From what I recall, the batteries did make a difference. I had commented a while back that my batteries were not fully charged (they were charged to 100% but the batteries were apparently not great and discharged super quickly). Once I got new batteries and fully charged those, everything worked.