sunfounder/SunFounder_PiCar-V #64: Client Hangs after connecting to RPi

Juanfdez11, 2021-02-06:
Hi, 
the client runs for me and when I enter the IP adress and default port it prompts me to the speed selection, where it freezes.
I have tried with por 7999 also but It wont work. The server on http://<RPi_IP_address>:8000/ does not work for me either.

Thanks i wait for some help.
[image]

Juanfdez11, 2021-02-06:
my bad i did not plug the power on smh

closed 2021-02-06 by Juanfdez11

weeurey, 2022-05-18:
What did you not plug in? Im facing the same issue

Juanfdez11, 2022-05-19:
> What did you not plug in? Im facing the same issue

if I'm not mistaken it was physically turning on the power on the extra circuit board. Double check that the rPi and the additional board are both being supplied power.