sunfounder/SunFounder_PiCar-V #65: ball track inop

mbar1, 2021-02-11:
i am unable to get the ball tracker to work.  
Traceback (most recent call last):
  File "C:\Users\LENOVO USER\Downloads\SunFounder_PiCar-V-master\SunFounder_PiCar-V-master\ball_track\ball_tracker.py", line 1, in <module>
    from picar import front_wheels, back_wheels
ModuleNotFoundError: No module named 'picar'
any suggestions?

Theosakamg, 2021-02-13:
You must run on the PI.
( picar module is installed by install_dependencies on PI )

mbar1, 2021-02-14:
Thanks, up and running.  Just not as well as I hoped for. Any code improvements available?

ctjoumas, 2021-04-30:
I was going to ask the same thing. I didn't have issues getting it to run, but it doesn't seem to work very well. Has anyone had success with this?