sunfounder/SunFounder_PiCar-V #69: Calibration fail : cali message received but nothing moves

andalousam, 2021-05-08:
Hi, the server started fine, and everything can moves, 

But netheir pyQT client nor web client can do the calibration 

The web client calibration part is even with more bugs

all cali command is doing nothing...

Log from the server when i try to calibrate front wheels

[08/May/2021 13:51:14] "GET /cali/?action=fwcali HTTP/1.1" 200 7670
"fwcali" command received
[08/May/2021 13:51:14] "GET /cali/?action=fwcali HTTP/1.1" 200 7670
"fwcalileft" command received
[08/May/2021 13:51:17] "GET /cali/?action=fwcalileft HTTP/1.1" 200 7670
"fwcalileft" command received
[08/May/2021 13:51:18] "GET /cali/?action=fwcalileft HTTP/1.1" 200 7670
"fwcalileft" command received
[08/May/2021 13:51:19] "GET /cali/?action=fwcalileft HTTP/1.1" 200 7670
"fwcalileft" command received
[08/May/2021 13:51:19] "GET /cali/?action=fwcalileft HTTP/1.1" 200 7670
"fwcalileft" command received

Please help, thats frustrating

andalousam, 2021-05-10:
Hi here is a little debug when sending cali request

Looks that some attribute issues in file front_wheels.py 

‐-------

Starting development server at http://0.0.0.0:8000/
Quit the server with CONTROL-C.
"fwcalileft" command received
Internal Server Error: /cali/
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/django/core/handlers/exception.py", line 35, in inner
    response = get_response(request)
  File "/usr/local/lib/python3.7/dist-packages/django/core/handlers/base.py", line 128, in _get_response
    response = self.process_exception_by_middleware(e, request)
  File "/usr/local/lib/python3.7/dist-packages/django/core/handlers/base.py", line 126, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
  File "/home/pi/SunFounder_PiCar-V/remote_control/remote_control/views.py", line 124, in cali
    fw.cali_left()
  File "/usr/local/lib/python3.7/dist-packages/SunFounder_PiCar-1.0.1-py3.7.egg/picar/front_wheels.py", line 133, in cali_left
    self.cali_turning_offset -= 1
AttributeError: 'Front_Wheels' object has no attribute 'cali_turning_offset'
[10/May/2021 11:32:36] "GET /cali/?action=fwcalileft HTTP/1.1" 500 62917

Chris-Schaefer-DE, 2021-08-25:
Howdy, I have the same problem and symptoms
- get the server messages
- but no servo moves 
- back wheel calibration does work.

_### [25/Aug/2021 09:39:08] "GET /cali/? HTTP/1.1" 200 7670
"camcali" command received
[25/Aug/2021 09:39:09] "GET /cali/?action=camcali HTTP/1.1" 200 7670
"fwcali" command received
[25/Aug/2021 09:39:09] "GET /cali/?action=fwcali HTTP/1.1" 200 7670
"bwcali" command received
[25/Aug/2021 09:39:09] "GET /cali/?action=bwcali HTTP/1.1" 200 7670
"camcaliup" command received
[25/Aug/2021 09:39:13] "GET /cali/?action=camcaliup HTTP/1.1" 200 7670
"camcaliup" command received
[25/Aug/2021 09:39:13] "GET /cali/?action=camcaliup HTTP/1.1" 200 7670
"camcaliup" command received
[25/Aug/2021 09:39:14] "GET /cali/?action=camcaliup HTTP/1.1" 200 7670
"camcaliup" command received_