sunfounder/SunFounder_PiCar-V #81: Migration Error when starting server.

vindekator, 2021-07-10:
System check identified no issues (0 silenced).

You have 1 unapplied migration(s). Your project may not work properly until you apply the migrations for app(s): auth.
Run 'python manage.py migrate' to apply them.

July 10, 2021 - 04:44:09
Django version 2.0, using settings 'remote_control.settings'
Starting development server at http://0.0.0.0:8000/
Quit the server with CONTROL-C.

======================================
when running the suggested command:
======================================

^C
pi@raspberrypi:~/SunFounder_PiCar-V/remote_control $ python manage.py migrate
Traceback (most recent call last):
  File "manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
ImportError: No module named django.core.management
pi@raspberrypi:~/SunFounder_PiCar-V/remote_control $

How do I fix this?

cfassbender, 2021-08-14:
python3 manage.py migrate

vindekator, 2021-08-15:
Thanks.  I meant to post a note that I saw that answer to someone else's question and I tried it.  That did the trick.  Just had to use the right version of python. 

Thanks again.

closed 2021-08-15 by vindekator