sunfounder/SunFounder_PiCar-V #84: The camera wont work.

Benjaminiscool, 2021-10-06:
I have done everything the instruction manual said, but the camera video feed won't work.

ShenhavLazarovich, 2021-12-25:
Have the same issue

amineremache, 2022-01-01:
Same issue, and it has been going for months, since July exactly, I tried to fix it myself, but did not have time for that, hopefully it will be fixed soon.

adrianadaca, 2022-02-05:
same problem here.

DarkTrick, 2022-04-15:
same problem here

waszak, 2022-08-04:
So I had this issue to.
So what I did first I went to "SunFounder_PiCar-V/mjpg-streamer".
I used command " sudo sh start.sh".  Got error also site "127.0.0.1:8080" didn;t work.

I found this fork of the mjpg-streamer. I replaced this current one and build with instruction.
https://github.com/jacksonliam/mjpg-streamer

After I installed I tried " sudo sh start.sh", webstite worked.

So I had to run again "sudo ./install_dependencies".

Then it started working for me.

This is my repo  with all the changes I did to make it work "https://github.com/waszak/SunFounder_PiCar-V"

If it helps anyone then probably then they should update library

liangste332, 2024-03-29:
I am also having trouble with getting the SunFounder PiCar front camera to work. I am interested in trying waszak's solution, but I am not sure how to install the new repo into the Raspberry Pi.

liangste332, 2024-03-29:
I updated the Raspberry Pi 3B+ code using waszak's github directory instead of sunfounder's. The servos for the camera work better than they did before, but the video feed still has nothing. Is it possible that I bent the FFC cable too much? Not sure what else I can do to troubleshoot.