sunfounder/SunFounder_PiCar-V #89: Client closes after connecting

derstange, 2022-04-08:
Hello,

I got a problem, connecting to the PiCar via Windows-Desktop Client.
After entering the IP and Port of the RasPi and loging in, the client opens for a second, freezes and shuts down.
The Server on the Raspi ist still running and the system check identified no issues.
By controlling the Car via Web-Client, there ist no problem logging in, but the Camera won't work.

[image]

[image]

[image]

[image]

balomerfaruk, 2022-05-19:
I have the same issues too.

derstange, 2022-05-19:
> Hi, did you find a fix for this? I have the same issue

Not yet, sorry :/

AUGSLAG, 2022-09-25:
Same issue. I think it may be a clock issue, the "qtimer" specifically.
But here are the errors. 
[image]

Gaotseun, 2023-03-31:
i have the same problem. was anyone able to solve this?

balomerfaruk, 2023-04-01:
Unfortunately. They didn’t respond me to fix this problem. 

iPhone’umdan gönderildi