sunfounder/SunFounder_PiCar-V #95: No module named picar_v (Dragit)

Parindraa, 2022-10-02:
pi@raspberrypi:/opt/SunFounder_Dragit/Dragit $ sudo ./start
Server running
Performing system checks...

Exception ignored in thread started by: <function check_errors.<locals>.wrapper at 0xb510c658>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/django/utils/autoreload.py", line 228, in wrapper
    fn(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/django/core/management/commands/runserver.py", line 124, in inner_run
    self.check(display_num_errors=True)
  File "/usr/local/lib/python3.9/dist-packages/django/core/management/base.py", line 356, in check
    all_issues = self._run_checks(
  File "/usr/local/lib/python3.9/dist-packages/django/core/management/base.py", line 346, in _run_checks
    return checks.run_checks(**kwargs)
  File "/usr/local/lib/python3.9/dist-packages/django/core/checks/registry.py", line 81, in run_checks
    new_errors = check(app_configs=app_configs)
  File "/usr/local/lib/python3.9/dist-packages/django/core/checks/urls.py", line 16, in check_url_config
    return check_resolver(resolver)
  File "/usr/local/lib/python3.9/dist-packages/django/core/checks/urls.py", line 26, in check_resolver
    return check_method()
  File "/usr/local/lib/python3.9/dist-packages/django/urls/resolvers.py", line 256, in check
    for pattern in self.url_patterns:
  File "/usr/local/lib/python3.9/dist-packages/django/utils/functional.py", line 35, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/usr/local/lib/python3.9/dist-packages/django/urls/resolvers.py", line 407, in url_patterns
    patterns = getattr(self.urlconf_module, "urlpatterns", self.urlconf_module)
  File "/usr/local/lib/python3.9/dist-packages/django/utils/functional.py", line 35, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/usr/local/lib/python3.9/dist-packages/django/urls/resolvers.py", line 400, in urlconf_module
    return import_module(self.urlconf_name)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/opt/SunFounder_Dragit/Dragit/Dragit/urls.py", line 18, in <module>
    from picar_v import views as picar_v_views
ModuleNotFoundError: No module named 'picar_v'

sunfounder, 2022-10-08:
Hi ,
Thanks for contacting us.
Could you please provide us your order number first? 
Which shopping platform did you get it?Amazon(US,CA,UK,Germany,etc...),Aliexpress,or elsewhere?
A short video or some pictures of the issues could be much more helpful.
Our staff will contact you later to help with that.
Best Regards,

***@***.***

Exception ignored in thread started by: <function check_errors..wrapper at 0xb510c658>
File "", line 1030, in _gcd_import
File "", line 1007, in _find_and_load
File "", line 986, in _find_and_load_unlocked
File "", line 680, in _load_unlocked
File "", line 790, in exec_module
File "", line 228, in _call_with_frames_removed

sunfounder, 2022-10-08:
Hi ,
Thanks for contacting us.
Could you please provide us your order number first? 
Which shopping platform did you get it?Amazon(US,CA,UK,Germany,etc...),Aliexpress,or elsewhere?
A short video or some pictures of the issues could be much more helpful.
Our staff will contact you later to help with that.
Best Regards,

***@***.***
 
From: Parindraa
Date: 2022-10-03 07:48
To: sunfounder/SunFounder_PiCar-V
CC: Subscribed
Subject: Re: [sunfounder/SunFounder_PiCar-V] No module named picar_v (Dragit) (Issue #95)
Closed #95 as completed.

sunfounder, 2022-10-08:
Hi ,
Thanks for contacting us.
Could you please provide us your order number first? 
Which shopping platform did you get it?Amazon(US,CA,UK,Germany,etc...),Aliexpress,or elsewhere?
A short video or some pictures of the issues could be much more helpful.
Our staff will contact you later to help with that.
Best Regards,

***@***.***
 
From: Parindraa
Date: 2022-10-03 07:48
To: sunfounder/SunFounder_PiCar-V
CC: Subscribed
Subject: Re: [sunfounder/SunFounder_PiCar-V] No module named picar_v (Dragit) (Issue #95)
Reopened #95.

Parindraa, 2023-02-05:
Hello, sorry for the late reply.
The order number is 112-3392157-5954664
I used Amazon to buy.
Here is the error,
I used the latest version of raspbian, I suspect your scripts and programs
are out of date.

***@***.***:/opt/SunFounder_Dragit/Dragit $ sudo ./start
Server running
Performing system checks...

Exception ignored in thread started by: <function check_errors..wrapper at
0xb510c658>
Traceback (most recent call last):
File "/usr/local/lib/python3.9/dist-packages/django/utils/autoreload.py",
line 228, in wrapper
fn(*args, **kwargs)
File
"/usr/local/lib/python3.9/dist-packages/django/core/management/commands/runserver.py",
line 124, in inner_run
self.check(display_num_errors=True)
File
"/usr/local/lib/python3.9/dist-packages/django/core/management/base.py",
line 356, in check
all_issues = self._run_checks(
File
"/usr/local/lib/python3.9/dist-packages/django/core/management/base.py",
line 346, in _run_checks
return checks.run_checks(**kwargs)
File
"/usr/local/lib/python3.9/dist-packages/django/core/checks/registry.py",
line 81, in run_checks
new_errors = check(app_configs=app_configs)
File "/usr/local/lib/python3.9/dist-packages/django/core/checks/urls.py",
line 16, in check_url_config
return check_resolver(resolver)
File "/usr/local/lib/python3.9/dist-packages/django/core/checks/urls.py",
line 26, in check_resolver
return check_method()
File "/usr/local/lib/python3.9/dist-packages/django/urls/resolvers.py",
line 256, in check
for pattern in self.url_patterns:
File "/usr/local/lib/python3.9/dist-packages/django/utils/functional.py",
line 35, in *get*
res = instance.*dict*[self.name] = self.func(instance)
File "/usr/local/lib/python3.9/dist-packages/django/urls/resolvers.py",
line 407, in url_patterns
patterns = getattr(self.urlconf_module, "urlpatterns", self.urlconf_module)
File "/usr/local/lib/python3.9/dist-packages/django/utils/functional.py",
line 35, in *get*
res = instance.*dict*[self.name] = self.func(instance)
File "/usr/local/lib/python3.9/dist-packages/django/urls/resolvers.py",
line 400, in urlconf_module
return import_module(self.urlconf_name)
File "/usr/lib/python3.9/importlib/*init*.py", line 127, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
File "", line 1030, in _gcd_import
File "", line 1007, in _find_and_load
File "", line 986, in _find_and_load_unlocked
File "", line 680, in _load_unlocked
File "", line 790, in exec_module
File "", line 228, in _call_with_frames_removed
File "/opt/SunFounder_Dragit/Dragit/Dragit/urls.py", line 18, in
from picar_v import views as picar_v_views
ModuleNotFoundError: No module named 'picar_v'

Parindraa, 2023-02-05:
Hello, sorry for the late reply.
The order number is 112-3392157-5954664
I used Amazon to buy.
Here is the error,
I used the latest version of raspbian, I suspect your scripts and programs
are out of date.

Exception ignored in thread started by: <function check_errors..wrapper at
0xb510c658>
Traceback (most recent call last):
File "/usr/local/lib/python3.9/dist-packages/django/utils/autoreload.py",
line 228, in wrapper
fn(*args, **kwargs)
File
"/usr/local/lib/python3.9/dist-packages/django/core/management/commands/runserver.py",
line 124, in inner_run
self.check(display_num_errors=True)
File
"/usr/local/lib/python3.9/dist-packages/django/core/management/base.py",
line 356, in check
all_issues = self._run_checks(
File
"/usr/local/lib/python3.9/dist-packages/django/core/management/base.py",
line 346, in _run_checks
return checks.run_checks(**kwargs)
File
"/usr/local/lib/python3.9/dist-packages/django/core/checks/registry.py",
line 81, in run_checks
new_errors = check(app_configs=app_configs)
File "/usr/local/lib/python3.9/dist-packages/django/core/checks/urls.py",
line 16, in check_url_config
return check_resolver(resolver)
File "/usr/local/lib/python3.9/dist-packages/django/core/checks/urls.py",
line 26, in check_resolver
return check_method()
File "/usr/local/lib/python3.9/dist-packages/django/urls/resolvers.py",
line 256, in check
for pattern in self.url_patterns:
File "/usr/local/lib/python3.9/dist-packages/django/utils/functional.py",
line 35, in *get*
res = instance.*dict*[self.name] = self.func(instance)
File "/usr/local/lib/python3.9/dist-packages/django/urls/resolvers.py",
line 407, in url_patterns
patterns = getattr(self.urlconf_module, "urlpatterns", self.urlconf_module)
File "/usr/local/lib/python3.9/dist-packages/django/utils/functional.py",
line 35, in *get*
res = instance.*dict*[self.name] = self.func(instance)
File "/usr/local/lib/python3.9/dist-packages/django/urls/resolvers.py",
line 400, in urlconf_module
return import_module(self.urlconf_name)
File "/usr/lib/python3.9/importlib/*init*.py", line 127, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
File "", line 1030, in _gcd_import
File "", line 1007, in _find_and_load
File "", line 986, in _find_and_load_unlocked
File "", line 680, in _load_unlocked
File "", line 790, in exec_module
File "", line 228, in _call_with_frames_removed
File "/opt/SunFounder_Dragit/Dragit/Dragit/urls.py", line 18, in
from picar_v import views as picar_v_views
ModuleNotFoundError: No module named 'picar_v'